deepnest-next/deepnest #11: 🐛 Overlapping parts

Dexus, 2024-12-23:
https://github.com/deepnest-io/Deepnest/issues/145

[image]
[bug-test-overlap.zip](https://github.com/user-attachments/files/18228099/bug-test-overlap.zip)

Dexus, 2025-01-31:
@ClockworkBastard is it possible that this only happends when the "Use rough approximation" is enabled?

I did run 30.000.000 tests now and I was now able to get one result that would overlap with the option enabled.

KoreUrala, 2025-02-23:
Hi! I just found this project out of having this exact issue!
Deepnest is a life saver for my sewing projects because it allows me to accommodate a myriad of different patterns in a single sheet, so first of all, thanks for your efforts!
Secondly, i'm having overlapping parts without the rough approximation feature :(

[image]

this happened while trying to fit five of these bad boys

[image]

in laser cut table (130cm * 90cm) dimensions.
I was using everything on default save for:
Part rotations - 2
Optimization type - Bounding box
Optimization ratio - 0.01
GA population - 256
GA mutation rate - 16

Using Windows 11 Pro running in an AMD Ryzen7 2700X along with a NVIDIA GeForce RTX 3060 with 16GB RAM
Dunno if the pc specs are relevant though...

Dexus, 2025-02-23:
@KoreUrala Thank you. Can you share the settings completely?  Is the SVG Preprocessor acativated?

KoreUrala, 2025-02-23:
SVG Prewhat...? Uhhh... idunno, it's probably on default...

Here, have a ss of the settings
[image]
Also this
[image]
Lemme run it again...
-_time skip_-

[image]
It infuriates me to no end...

KoreUrala, 2025-02-25:
Yeah, still happens with normalizer, I also noticed it behaves slightly better in inches than in millimeters... very slightly...

Dexus, 2025-02-25:
Thank you for letting me know that. Hope that I can find the "bad" part now,

Dexus, 2025-02-27:
Thank you @KoreUrala! I have now been successful in recreating the behaviour and am currently finalising the fix. Will be released with v1.4.1. I'm running some tests now, and if there are no more overlaps, it will be released tonight or tomorrow during the day.

commit f99db97a (2025-02-27): overlapping polygons on results fixes #11
commit 0f033d18 (2025-02-27): overlapping polygons on results fixes #11
commit 0da71fb6 (2025-02-28): overlapping polygons on results fixes #11
PR #105 (2025-04-23): Fix: overlapping and nesting in holes
This PR fixes #11 and fixes #99.

I have rewritten the entire part placement and added a more robust check for the overlapping parts.

After several error correction loops I am happy with the result. V2 should get a propper rewrite and refactoring.

Close older PRs which where not as good as I wished.
Close #100 
Close #101
closed 2025-02-28 by Dexus