deepnest-next/deepnest #25: Hello, I would like to ask how to make it so that after arranging, each one shows the outer bounding box instead of just showing the last bounding box, and also that it needs to export the bounding boxes as well.

pofenx, 2025-01-07:
Hello, I would like to ask how to make it so that after arranging, each one shows the outer bounding box instead of just showing the last bounding box, and also that it needs to export the bounding boxes as well.
[image]

Dexus, 2025-01-07:
enhancement: have a option to allow the export with/without borders.

pofenx, 2025-01-07:
Will this feature be added under the main branch later, or is this feature already available in another branch now?

Dexus, 2025-01-07:
> > enhancement: have a option to allow the export with/without borders.
> 
> Will this feature be added under the main branch later, or is this feature already available in another branch now?

I will check, I think that I will fix it already on the main branch and add it also to v2.

Dexus, 2025-01-07:
@pofenx  @Toku-loveStudy

It will be in v1.3.1 that I will release later today maybe around 20:00 CET.

pofenx, 2025-01-07:
> @pofenx @Toku-loveStudy
> 
> It will be in v1.3.1 that I will release later today maybe around 20:00 CET.

Thank you very much! However, there is still a problem: the exported SVG or DXF files still do not have borders. Is it possible to include the boundary box when exporting the formatted SVG or DXF files?
[image]
[image]

Dexus, 2025-01-07:
I was not releasing it yet or did I? I'm working on it - but have to do first payed stuff.

commit 0cf323fe (2025-01-07): Possible fix with sheetid + boarders

fixes #25
fixes #13
commit 228dff1e (2025-01-07): Fix boarder and double sheet-ids. Add new Options.

fixes #25
fixes #13

Add config for Export Sheets Bound-Boarders

real fix for #25 and #13

The real Fix!

Fix Export
PR #27 (2025-01-07): Feat/show boarders on each bin issue for #25 and fix for #13
This PR fixes #25 and fixes #13.

Also it adds a new option export with/out boarders.
commit 253f6640 (2025-01-07): Fix boarder and double sheet-ids. Add new Options.

fixes #25
fixes #13

Add config for Export Sheets Bound-Boarders

real fix for #25 and #13

The real Fix!

Fix Export
closed 2025-01-07 by Dexus