deepnest-next/deepnest #33: Can the total space used and total space of each layout result be used as utilization, and can the utilization of each layout result be calculated and displayed?

pofenx, 2025-01-14:
Can the total space used and total space of each layout result be used as utilization, and can the utilization of each layout result be calculated and displayed?
$$\[
text{Utilization} = \left( \frac{\text{Used Space}}{\text{Total Space}} \right) \times 100\%
\]$$

### Explanation of Terms

- **Used Space**: The amount of space that has already been utilized.
- **Total Space**: The maximum available space.``

pofenx, 2025-01-14:
[image]

Dexus, 2025-01-14:
This is calling fitness, in deepnest. Because we have not only one Bin, we support multiple Bins, and have to calculate it a bit different. I don't see why we should display it, becuase its only added to the "results" if it is better then the best in the currently showed list of 10 results.

closed 2025-01-15 by Dexus