deepnest-next/deepnest #36: Display widget material text information

pofenx, 2025-01-18:
Hello, if there are many small parts and there is text numbering information for the small parts in the DXF file or converted SVG file, how can we also display the descriptive text information that differentiates the text and the small parts inside, so that the exported file also contains text information?

[image]

Dexus, 2025-01-18:
This is not possible. Only way would be make those information a path, that would then always in the item, also after the nesting, but it will not keep the color or be on a different layer. It will be then part of the item and would need a cleanup afterwards.

pofenx, 2025-01-18:
Thank you for your answer.

closed 2025-01-19 by Dexus

pofenx, 2025-03-01:
I did this, parsing the file name as its material information, and it served as a configuration item to determine whether it is enabled.,coding changed is :

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Dexus, 2025-03-01:
@pofenx Thank you.

Could you provide a PR for this?

pofenx, 2025-03-01:
ok ,I have submitted the PR

Dexus, 2025-03-01:
@pofenx I'm not sure, your PR is only a edited readme.md

pofenx, 2025-03-02:
Sorry, there might be an initialization error, I have resubmitted it.