deepnest-next/deepnest #43: 1.4.0 Nesting doesnt work

tarman3, 2025-02-03:
Tried new version 1.4.0 from release page and build by npm
Result is empty after push **Start nest**

1.3.1 working fine

https://github.com/user-attachments/assets/10713934-50fe-4294-bddc-00054505a34c

Dexus, 2025-02-03:
Ok, i think it has todo with the build of the release on MacOS.
Will rebuild it on a Linux System

tarman3, 2025-02-03:
Strange. Downloaded updated release, but nothing change on my side

> Please make sure that chrome-sandbox need to be root owned and 4755 mode.
Please write about this little bit more?

Dexus, 2025-02-03:
Please run the deepnest-v1.4.0 binary from the command line, this should show you the message

Dexus, 2025-02-03:
It should show something like:

`The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /home/......./chrome-sandbox is owned by root and has mode 4755.`

tarman3, 2025-02-03:
Sorry
Probably I`m doing something wrong
Looks like now it is working

Thank you

closed 2025-02-03 by Dexus

Dexus, 2025-02-04:
@tarman3 https://discussions.apple.com/thread/253714860?sortBy=best ?

Was it the file flagged? xattr?

Dexus, 2025-02-04:
Okay sorry did not read it again. Had some reports on different channels