deepnest-next/deepnest #45: Need Apple Developer Account - Financial support needed.

Dexus, 2025-02-04:
We are currently running the MacOS release in the gatekeeper, the problem can be solved as described here at https://discussions.apple.com/thread/253714860?sortBy=best.

To be able to sign all the apps in the future, I need an Apple Developer Account. This costs me 99€ net/gross? - If you are using MacOS, I would be grateful if you would make a donation via https://www.patreon.com/c/deepnest_next.

With this you not only support the development for MacOS but for deepnest-next in general and all deepnest developments that are OpenSource. 

Thank you very much!

beckmx, 2025-02-21:
if i want to build on osx, do i need visual studio? or just by building with node is enough?

Dexus, 2025-02-21:
> if i want to build on osx, do i need visual studio? or just by building with node is enough?

As far as I can tell, there is no Visual Studio installed on the online Mac provided to me. I only have xcode, nodejs + npm/yarn installed + maybe a few other brews.

Dexus, 2025-04-04:
I have now a Apple Account, and have build a new system for release. Thanks to all the supporters!

closed 2025-04-04 by Dexus