deepnest-next/deepnest #46: :bug: <conversion server>

MasterZahner, 2025-02-05:
**Describe the bug**
it happens again and again that the import of dxf does not work. because the message could not contact file conversion server appears. what exactly is the reason that this happens from time to time. i have already invested a lot of time in my own version of deepnest and now this error appears.

**Expected behavior**
Please explain to me what this conversion server does and where it accesses and whether this cannot be circumvented..

**Screenshots**

[image]

Dexus, 2025-02-06:
I will close this issue if no reply is done within 3 days.

pofenx, 2025-03-01:
Hello, I used Inkscape in Linux to convert SVG to DXF format. When using the Python libraries ezdxf and svgwrite to convert DXF back to SVG format, the results were quite good. You can give it a try. The conversion from SVG to DXF involves first using the Inkscape software to convert SVG to EPS, and then using the pstoedit software to convert EPS to DXF. Below is my code for your reference.
You need to install pstoedit and inkscape in Linux, the commands are yum install pstoedit, yum install inkscape.
def svg_to_eps(input_file):
    eps_file = os.path.splitext(input_file)[0] + ".eps"
    subprocess.run(["inkscape", input_file, "--export-eps", eps_file], check=True)
    return eps_file

def svg_to_dxf(input_file):
    eps_file = svg_to_eps(input_file)
    dxf_file = os.path.splitext(input_file)[0] + ".dxf"
    subprocess.run(["pstoedit", "-dt", "-f", "dxf:-polyaslines -mm", eps_file, dxf_file], check=True)
    os.remove(eps_file)  # Optional: Remove the EPS file after conversion
    return dxf_file

use : dxf_file = svg_to_dxf(temp_svg_path) 

dxf2svg code is ：

def dxf_to_svg(dxf_filename, svg_filename):
    result = {
        "width": "",
        "length": "",
        "thickness": "",
        "partNumber": "",
        "material": "",
        "quantity": "",
        "specification": ""
    }

    # Open DXF file
    doc = ezdxf.readfile(dxf_filename)
    msp = doc.modelspace()

    # Calculate model space extents
    extents = bbox.extents(msp)
    min_x, min_y, min_z = extents.extmin
    max_x, max_y, max_z = extents.extmax

    # Create SVG file with viewBox
    width = max_x - min_x
    height = max_y - min_y

    result["width"] = max_y
    result['length'] = max_x

    svg_document = svgwrite.Drawing(filename=svg_filename,
                                    viewBox=f"{min_x} {-max_y} {width} {height}")

    def bulge_to_arc(start_x, start_y, end_x, end_y, bulge):
        # Convert bulge to arc parameters
        dx = end_x - start_x
        dy = end_y - start_y
        chord = math.sqrt(dx * dx + dy * dy)

        # Calculate arc parameters
        alpha = 4 * math.atan(abs(bulge))
        radius = chord / (2 * math.sin(alpha / 2))

        # Determine sweep and large-arc flags
        sweep_flag = 1 if bulge < 0 else 0  # Modify here: bulge < 0 means concave curve, sweep_flag should be 1
        large_arc_flag = 1 if abs(alpha) > math.pi else 0

        return radius, large_arc_flag, sweep_flag
    

    # Process entities
    for entity in msp:
        if entity.dxftype() == 'POLYLINE':
            vertices = list(entity.vertices)
            if len(vertices) == 0:
[... 62 lines cut ...]

                result['quantity'] = parts[0]  # Quantity
                result['material'] = parts[2]  # Material
                result['thickness'] = numbers[0]  # Thickness
                result['specification'] = f"{parts[1]}*{result['width']}"  # Specification
            else:
                result['partNumber'] = text  # Part number

            # Add text with proper color based on layer
            text_element = svg_document.text(text,
                                             insert=(x, y),
                                             font_size=height)

            if entity.dxf.layer == "SCRIBE":
                text_element.fill("green")
            elif entity.dxf.layer == "TEXT":
                text_element.fill("blue")
            else:
                text_element.fill("black")

            svg_document.add(text_element)

    # Save SVG file
    svg_document.save()
    return result