deepnest-next/deepnest #47: 1.4.0 x64 on Linuux Mint - blank grey screen while nesting

baldyman01, 2025-02-23:
On Linux Mint 21 Deepnest 1.4.0 x64 stalls at the nesting screen, which remains a blank grey screen. When manually stopped no export option is offered indicating that it has indeed been doing nothing, not just that I can't see it.

1. Download deepnest-v1.4.0-linux-x64,zip and unzip into a folder 
2. Double clicked the deepnest-v1.4.0 file (which already has executable permission)
3. Imported simple test SVG
4. Tick the check box for my stock material
5. Press start nest

Result -

[image]

It remains this way indefinitely.

Test svg 

[image]

OS -

NAME="Linux Mint"
VERSION="21.3 (Virginia)"
ID=linuxmint
ID_LIKE="ubuntu debian"
PRETTY_NAME="Linux Mint 21.3"
VERSION_ID="21.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=virginia
UBUNTU_CODENAME=jammy

NVIDIA Driver Version: 535.183.01

baldyman01, 2025-02-23:
FYI I just followed the npm install instructions and that works exactly as expected. 

The only problem with this is I have singularly failed to figure out how to run this from a desktop shortcut or menu entry so have to run it from the CLI. Otherwise its all good.

Dexus, 2025-02-23:
Once you run this from cli, does it give you an error?

Please also test it with a run where `deepnest_debug=1` environment is set. And check what the debug windows shows you in the console like you may know from chromium browser.

baldyman01, 2025-02-23:
This appears on application startup -

`pete@MONSTER:~/Applications/deepnest-v1.4.0-linux-x64$ ./deepnest-v1.4.0 
[79879:0223/135210.982529:ERROR:CONSOLE(1)] "Request Autofill.enable failed. {"code":-32601,"message":"'Autofill.enable' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[79879:0223/135210.982644:ERROR:CONSOLE(1)] "Request Autofill.setAddresses failed. {"code":-32601,"message":"'Autofill.setAddresses' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[79879:0223/135211.537118:ERROR:CONSOLE(1)] "Request Autofill.enable failed. {"code":-32601,"message":"'Autofill.enable' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[79879:0223/135211.537172:ERROR:CONSOLE(1)] "Request Autofill.setAddresses failed. {"code":-32601,"message":"'Autofill.setAddresses' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)`

On hitting Start Nesting this appears -

`starting background!`

On stopping it this appears -

`stopped! [
  BrowserWindow {
    setBounds: [Function (anonymous)],
    _events: [Object: null prototype] {
      blur: [Function (anonymous)],
      focus: [Function (anonymous)],
      show: [Function: visibilityChanged],
      hide: [Function: visibilityChanged],
      minimize: [Function: visibilityChanged],
      maximize: [Function: visibilityChanged],
      restore: [Function: visibilityChanged],
      closed: [Function (anonymous)],
      'ready-to-show': [Function]
    },
    _eventsCount: 9,
    _browserViews: [],
    devToolsWebContents: [Getter]
  }
]
[79879:0223/135309.095209:ERROR:CONSOLE(1)] "Request Autofill.enable failed. {"code":-32601,"message":"'Autofill.enable' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[79879:0223/135309.095266:ERROR:CONSOLE(1)] "Request Autofill.setAddresses failed. {"code":-32601,"message":"'Autofill.setAddresses' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)`

Dexus, 2025-02-23:
Okay, so it's no bug or problem with the deepnest app, else there should be shown something.

How is your system setup? CPU, Cors, RAM? How long did you let the nesting run? An how are your settings?

baldyman01, 2025-02-23:
System : 16 core AMD Threadripper 2950, 64GB Ram. I could do with being able to use more than 8 out of 32 threads btw :) 

I can't time the first nest time for this test file as by the time the animated screen wipe has started it's already done. For reference, on the npm installed version this work sheet takes about 5 seconds to first solution -

[image]

Dexus, 2025-02-23:
Ok, so I think it has todo with the different libs and systems. Linux is not like Linux. And the provided distro Packages can be different versions.

I will check if I can provide Flatpack and/or Snap packages. Think this would be the best way.

baldyman01, 2025-03-31:
Just FYI, 1.5.1 has the same, if you didn't already know that.

skylarmt, 2025-08-10:
Having the same issue, 1.5.6 on Kubuntu 22.04

aerospaceindonesian-boop, 2025-08-11:
i got same issue, 1.5.6 on windows 11

Dexus, 2025-08-11:
@skylarmt @pratamakurniansyah-netizen please run 
```sh
export deepnest_debug=1
./deepnest-v1.5.6
```

- It opens one additional developer console, make it a own window (right ... menu)
- close the app
- restart the app now 2 additional developer consoles must be open
- try your files, if one of the windows showing errors in the console or it showing "...crashed..." as overlay on the background titled Developer Console give me the details.

I'm unable to check all systems