deepnest-next/deepnest #65: 1.4.0/1.5.0 Nesting doesnt work for windows

yada886, 2025-03-04:
Tried new version 1.4.0 and 1.5.0 from release page and build.
Result is empty after push Start nest

[image]

[image]

 - OS: Windows11 Enterprise
 - deepnest Version: 1.4.0/1.5.0

Dexus, 2025-03-04:
Please open a powershell and type: `$env:deepnest_debug=1` in the folder where you put deepnest. then start the app with deepnest-v1.5.0.exe you will get some output to the console, try to start the nesting - please not press stop after it.

Then you also get 2 debug windows, please copy the content form the console tab. So I can get more infos to whats wrong.

yada886, 2025-03-05:
Thank you for your reply. I will try to get more information. However, before importing the DXF file, I'm experiencing a conversion problem. Could you please give me some advice on how to solve this issue?

[image]

Dexus, 2025-03-05:
You have the option to export your DXF directly as SVG. I am currently still working on a usable converter. The current online converter is still from the original author.

yada886, 2025-03-06:
Thank you for the advice. I obtained the information using debag mode and downloaded files from three consoles. Does this meet your requirements?

[-1741225357444.log](https://github.com/user-attachments/files/19099810/-1741225357444.log)
[-1741225362529.log](https://github.com/user-attachments/files/19099812/-1741225362529.log)
[-1741225366686.log](https://github.com/user-attachments/files/19099811/-1741225366686.log)

[image]

Dexus, 2025-03-06:
@yada886 more or less, I need the out put from the powershell when the programm is hanging. and the output ob the background.html via debugmode enabled. all I see is only the index.html log - three times.

A short info about your settings (screenshot) and your system would also be helpful.

Dexus, 2025-03-06:
Like:

[image]

RED: the PowerShell console
BLUE: The WebConsole for index.html
GREEN: the WebConsole for background.html

All three outputs are needed.

And your current system info and program settings. I think its nothing with the programm - except you have changed something.

yada886, 2025-03-06:
Thank you. I imported a DXF file and started nesting.
I have obtained the following: 
RED: the PowerShell console
BLUE: The WebConsole for index.html

However, I couldn't find the backgraound.html. How can I access it?
GREEN: the WebConsole for background.html

Sorry to bother you.

[image]

[image]

Dexus, 2025-03-06:
> However, I couldn't find the backgraound.html. How can I access it?
> GREEN: the WebConsole for background.html

make the DevTool a own window and restart the app. That should work to bring the background DevTools as new window.

yada886, 2025-03-06:
Thank you. I have attached the condition of the three consoles after starting the nesting process.
Also, I have attached the setting information.

[image]

[image]

[image]

Dexus, 2025-03-06:
And it worked with a version before 1.4.0? If so, which version and downloaded from this projekt or an other one? I can't see currently an Issue. I have also checked it here on different Win10/11 systems. It works everything here.

Please let me know your System Setup:
- Windows 11 Enterprise, howmany cores howmuch ram?

yada886, 2025-03-06:
v1.4.0 has same issue. I used v1.0.5 before. It is working.
Windows11 Enterprise
Core:8 core
RAM:32GB

Dexus, 2025-03-07:
Sorry I can't see what should be wrong. There is no error output. Ich will try to add more debug infos, so we can check what the problem is.

yada886, 2025-03-10:
Thank you for your response. I look forward to being able to use it in the upcoming update.

Dexus, 2025-03-11:
@yada886 please make a video in debug mode and let it run for around 5 minutes, please also show your taskmanager and the CPU/RAM load. Where the nesting is hanging.

closed 2025-03-22 by Dexus