deepnest-next/deepnest #66: [Feature] Maintain true curved lines

HuidoLtd, 2025-03-07:
As you may know, deepest doesn't currently preserve true curves when doing the nesting but it breaks them down into straight segments. That's a problem if making something accurate that should have round holes, curves etc. in the parts.

So the suggestion for this "problem" is that the software should keep the original shape in memory while making the nesting with simplified (enough) copies for the algorithm. Then when exporting the results, the app should use the original shapes instead to maintain the curves.

How's that sounding to you?

Dexus, 2025-03-07:
I think that should be our goal, but we need to know, that once you convert between DXF and SVG you already lost much informations - based on the converter. Sure if we can use commercial thirdparty tools, we would have here no problem. BUT: We have here a open source software with very old code base, that first needs to modernized. Any maybe we find other coders, that would help to process the transformation.

willmurnane, 2025-03-19:
Oh man, I'm so happy to find this project revived. I found Deepnest back at version 1.0.5, used it for a year or so, got frustrated with its limitations, started a clean sheet rewrite, and got way bogged down into the details. I have a bunch of code related to this concept (and #40 by the looks of it), I should clean that up and try to push it up.

brunomunizaf, 2025-07-25:
Oh man, I'd love to see that go further @willmurnane and @Dexus !! This project is simply too amazing!! Thanks a lot for reviving it