deepnest-next/deepnest #71: :bug: File Conversion Server Down

anuragrao04, 2025-03-21:
**Describe the bug**

Importing as dxf raises an error - File Conversion Server Not Reachable

**To Reproduce**

Steps to reproduce the behavior:

1. Try importing a dxf
2. See error

**Expected behavior**

The dxf file should be imported after file conversion

**Desktop (please complete the following information):**

 - OS: MacOS, Linux, Windows
 - Version: Latest development build as of march 22nd 2025

**Additional request**

It would be awesome if we can open source the file conversion server so we can host our own
**DEBUG information**

Open a new shell within the deepnest app folder.
Set a temporary environment `deepnest_debug=1` on windowns you can type: `$env:deepnest_debug=1` on mac and linux `export deepnest_debug=1` this should open 2 debug windows when you start now the `deepnest-v*.*.*` binary within the shell. Please provide the content from the shell and also the content from the `console` tab of the debug windows. please note, the debug windows for the background will go if you press `stop` on the deepnest nesting gui.

anuragrao04, 2025-03-21:
## Additional Information:

The file conversion server at `https://convert.deepnest.io` returns an error 522. It means that cloudflare is able to contact the backend server but it took too long to respond. It might be some kind of resource hogging process,

Dexus, 2025-03-22:
Its not our server or service!

Dexus, 2025-03-22:
I've just tried to get a clean setup for the converter, but unfortunately I'm very tired now and have to sleep. 

```
%APPDATA% on Windows
$XDG_CONFIG_HOME or ~/.config on Linux
~/Library/Application Support on macOS
```

then goto the `deepnest` folder and edit the `settings.json`

the line should look like `"conversionServer": "https://converter.deepnest.app/convert",`

This will be the URL so that you can currently use our server. I have not yet tested it via API and only default settings are active. Anyone who wants to test it can do so right now.

Werde nach einem Nap weiter an dem Problem arbeiten.

_If it's worth it to you, I'm happy about sponsors here on Github or Patreon - you benefit from the API - I don't_

Dexus, 2025-03-22:
Quick update before I really go to bed. API is running, unfortunately sometimes with errors, then you have to try again. I will look at the logs later. But my tests have worked at least so far, whether the previous settings still fit for DXF imports and exports, I can't verify right now.

anuragrao04, 2025-03-22:
Thanks a bunch for the super quick replies and fixes. I will try this out locally. Do you plan on maintaining `converter.deepnest.app` long term? 

It is worth it to me but I am not in a position to sponsor the project. We are a student run university club making RC aircrafts in a third world country. 

However, we can offer hosting in terms of the free credits we receive from Github student pack, and other student benefits. It's a short term solution but it's the most support I can think of. 

Deepnest is a great tool and I was really surprised to see it's the only free option available beside paid proprietary stuff from autodesk. I'll try to get some PRs in to the repository!

On a friendlier note, you sound burnt out. I hope you feel better

Dexus, 2025-03-22:
Hi, thanks for your reply. 

Yes, I am planning to maintain everything in the long term.

It's okay - it was just a friendly question, but it's just used a lot commercially - at least my guess without getting anything in return. But anyway, it's like many hobbies, they just cost a lot of money 💯

 I think there is other software out there that is open source, but I have to admit I've tried a lot and seen a lot of half hearted stuff.

Thanks, yes I'm good, was a hard week and a very long day yesterday.

anuragrao04, 2025-03-23:
The official distribution for Macos (ARM) does not work on MacOS. It gives a `The application “deepnest-v1.5.0” can’t be opened.` error and does not launch. I'm not sure where to find further logs to debug this. 

I'm running the application by building from this repository. It looks like the new conversion server returns a status code 500 for my dxf files. I'm not able to figure out why this happens since we can't self host the conversion server yet. Would you please check the conversion server logs to see what happened?

anuragrao04, 2025-03-23:
Here is the response from the file conversion server (`https://converter.deepnest.app/convert`)

```json
{
    "error": "1B2 error: B2 Error: \u003cnil\u003e\nHTTP/1.1 401 \r\nContent-Length: 68\r\nCache-Control: max-age=0, no-cache, no-store\r\nConnection: keep-alive\r\nContent-Type: application/json;charset=utf-8\r\nDate: Sun, 23 Mar 2025 09:03:15 GMT\r\nServer: nginx\r\nStrict-Transport-Security: max-age=63072000\r\n\r\n{\n  \"code\": \"expired_auth_token\",\n  \"message\": \"\",\n  \"status\": 401\n}"
}
```

anuragrao04, 2025-03-23:
The same response cleaned up and beautified:

```json
{
    "error": "1B2 error: B2 Error: <nil>
    HTTP/1.1 401 
    Content-Length: 68
    Cache-Control: max-age=0, no-cache, no-store
    Connection: keep-alive
    Content-Type: application/json;charset=utf-8
    Date: Sun, 23 Mar 2025 09:03:15 GMT
    Server: nginx
    Strict-Transport-Security: max-age=63072000
    {
    "code": "expired_auth_token",
    "message": "",
    "status": 401
    }"
}

```

anuragrao04, 2025-03-23:
What is 1B2? Can't find anything about it online

Dexus, 2025-03-23:
It's my my backend.

Dexus, 2025-03-23:
> The official distribution for MacOS (ARM) does not work. It gives a `The application “deepnest-v1.5.0” can’t be opened.` error and does not launch. I'm not sure where to find further logs to debug this.
See https://github.com/deepnest-next/deepnest/issues/45

Dexus, 2025-03-23:
Release will comming soon.

Dexus, 2025-03-25:
### Information:

We found a bug within our new converter, thats why I was able to test it local which works, but the public prod was not working. I have now fixed it.

Dexus, 2025-03-25:
Release v1.5.1 includes this fix.

closed 2025-03-25 by Dexus