deepnest-next/deepnest #73: The unit is still incorrect.

Tigger86, 2025-03-25:
i'm writing here again to report that in the current version 1.5.1, there is still an issue with the unit conversion. I have already discussed this topic [here.](https://github.com/Jack000/Deepnest/issues/196#issuecomment-2751738824) 100 mm are not correctly output as 100 mm.

Dexus, 2025-03-25:
The modest thing is that every graphics programme works differently.

AutoCAD used to include a unit, but that's probably no longer the case, it's just "points". With Inkscape, the resolution and other things play a role. I'll have to test Fusion360 again. Affinity Designer, I also have to test it again, at least it opens everything correctly with the 96dpi, both when exporting and importing.

It could all be so easy 🗡

Tigger86, 2025-03-25:
i use fusion 360 and lightburn... but lightburn just for svg`s

trivster, 2025-03-26:
The settings in JSON are confusing!

Here are my JSON settings to make DXFs import correctly in mm.
I have 'space between parts' set as 4mm (15.118 x 25.4mm / 96 dpi = 4mm) because I cut using waterjet with very narrow kerf.

First set units, scale (dpi), dxf import scale & dxf export scale in the JSON file (find by typing %appdata% into the file explorer window).
Then change values like part spacing, tolerances, optimzation type etc. in the Deepnest app.

Interstingly, the 'dxf import units' dropdown box in the app is blank?!

{
  "units": "mm",
  "scale": "96",
  "spacing": 15.118110236220474,
  "curveTolerance": 0.03779527559055119,
  "rotations": 4,
  "threads": 4,
  "populationSize": 10,
  "mutationRate": 10,
  "placementType": "gravity",
  "mergeLines": false,
  "timeRatio": 0.5,
  "simplify": false,
  "dxfImportScale": "25.4",
  "dxfExportScale": "25.4",
  "endpointTolerance": 0.03779527559055119,
  "conversionServer": "https://converter.deepnest.app/convert",
  "useSvgPreProcessor": false,
  "useQuantityFromFileName": false,
  "exportWithSheetBoundboarders": true,
  "exportWithSheetsSpace": true,
  "exportWithSheetsSpaceValue": 1889.7637795275593
}

Tigger86, 2025-03-26:
thanks, i ll change the json file tonight and test it. hopefully the server is not down anymore ?! last night it was. 
i will see if the dropdownbox was empty. Thanks

Tigger86, 2025-03-26:
okay, I'm waiting for your help

trivster, 2025-03-26:
I had to change the settings in JSON because the dxf import scale was not correct after changing from .io to .net deepnest software.
It sounds like the OP had the same problem. Changing the settings and units only in the app did not work for me.

Dexus, 2025-03-26:
> I had to change the settings in JSON because the dxf import scale was not correct after changing from .io to .net deepnest software. It sounds like the OP had the same problem. Changing the settings and units only in the app did not work for me.

Yeah, we don't use the same software, that the cause. But the scale would here complete wrong, which would affect some math.

trivster, 2025-03-26:
Is that why the export nest to dxf is not working for me? https://github.com/deepnest-next/deepnest/issues/75
What is the solution? Clean install of 1.5.1 after uninstalling all versions and deleting the deepnest folder in appdata?

trivster, 2025-03-26:
@Tigger86 can you export nest to dxf (even if the part scale is wrong)?

Tigger86, 2025-03-26:
> [Tigger86](https://github.com/Tigger86) can you export nest to dxf (even if the part scale is wrong)?

jap, its working..  i tried it yesterday

trivster, 2025-03-26:
Hi
I have reverted to default settings.
Here are screenshots of the scale problem. The dxf part should be 600mm long. First image units are inches, second is mm.

I have tried all the different dxf import units (points, picas, inches, mm, cm) but the scale is wrong on all of them. This is why I had to change the dxf scale in JSON.

[image]

[image]

Dexus, 2025-03-26:
I really don't understand it... why was this scale installed back then? Was it really the converter or were there other reasons, we'll probably never know.

All in all, I can now say that we don't actually need the scale. If we leave the current settings for the DXF import at Point and set Export to Inch, we get the correct values.

At least I have not found any necessary change, except that I have to change the default export to inch. So that the import of 100mm is in mm view 100mm and on export the 100mm are 100mm.

Dexus, 2025-03-26:
From my point of view, I see no need to do anything. I have worked with the data available to me and all editors and AutoCAD 2025 say the same thing, the values with 

import = point
Export = Inch 

result in the correct values, regardless of what the SVG DPI says as unit/inch. But I can imagine that it is related to images, i.e. images for engraving?

trivster, 2025-03-26:
Yes, that now works OK
It is counter-intuitive though - if you want the export to be in mm set to inches?!

Thanks for your help. I have become a sponsor as the software is very useful for my business.

Dexus, 2025-03-26:
Yes, it's a bit crazy, I need to take another look at it, but then in connection with raster images, because I think it was once intended to nest images in SVGs for laser cutting.

Tigger86, 2025-03-27:
Maybe i did it wrong or so.. But i have still a problem with the units.. i created a dxf file of 100 x100 mm in fusion 360 and then i put it into estlcam... and the dxf is now 7200 mm ?! did i forget to set something ? i downloaded the last version again... sorry

Dexus, 2025-03-27:
F360 100mm dxf to estlcam = 7200mm?
Or f360 dxf> deepnest dxf> estlcam =7200mm? If so, export as point for estlcam or estlcam setting?

Kannst auch gerne auf Deutsch schreiben.

Tigger86, 2025-03-27:
Was meinst du mir estlcam settings ?  Also wie gesagt . Ich habe eine DXF von f360 gezogen und diese dann in deepnest importiert.die maße stimmen dann soweit. 100x100. Wenn ich diese dann exportiere als DXF und in estlcam öffne, habe ich utopische Werte.  Was ist denn Points ? SVG.dxf oder json.

Tigger86, 2025-03-28:
Jap, daran lag es.. das hab ich übersehen. Danke dafür.

Tigger86, 2025-03-28:
solangsam komm ich mir nervig vor.. tut mir leid..aber ich versuche gerade etwas zu verschachteln und dabei ist mir was aufgefallen.. eigentlich soll in der ersten dxf ein loch sein.. der wird in deepnest garnicht angezeigt.. woran kann das liegen ?

[image]
[image]
[image]

Dexus, 2025-03-28:
Womit Zeichnest du? sollte es Inkscape oder dergleichen sein, bitte alle Objekte in Pfade konvertieren oder aber du aktivierst den SVG Preprocessor in den Einstellungen. Gerne mal die DXF hochladen, falls das damit nicht behoben werden kann.

Tigger86, 2025-03-29:
ich erstelle die dxf`s meistens mit f360.  ich hab danach sogar mal versucht in Lightburn die datei zu gruppieren.. und wieder als dxf zu exportieren.. aber deepnest erkennt einfach den kreis nicht. :-)

Dexus, 2025-03-29:
> ich erstelle die dxf`s meistens mit f360.  ich hab danach sogar mal versucht in Lightburn die datei zu gruppieren.. und wieder als dxf zu exportieren.. aber deepnest erkennt einfach den kreis nicht. :-)  
> 
> 
> [dxf](https://we.tl/t-nszSIbErKL)

Jetzt mal im ernst, hast du unter Einstellungen mal die svg preprocessor aktiviert? Und dann die dxf importiert?

Tigger86, 2025-03-29:
Ja sorry.. Nein, hatte ich nicht aktiviert, weil ich dachte... es ist ja ne dxf.. dachte svg. hat damit nichts zu tun..  Sorry. für deine Zeit..

🙈

Tigger86, 2025-03-29:
ich muss nochmal dumm fragen.. aber warum passieren diese überlappungen ? eingestellt sind 6 mm abstand .
muss ich noch was in der einstellung ändern ?

[image]
[image]
[image]

Dexus, 2025-03-29:
Eine gute Frage. Meistens kann man es beim erneuten starten des nesten umgehen. Problem wurde schon mehrfach angegangen, aber da man einmal etwas behebt, kommt es dann doch bei anderen wieder vor. #11 #53

Tigger86, 2025-03-29:
okay,, so langsam komm ich mir selber dumm vor.. aber ich bekomm es nicht hin.. folgende dxf.. 255 breit und 420.680 lang..  ich habe dne import auf points stehen.. export zu inches.. aber die maße in deepnest sind dabei dann andere : 344.8 und 568.8 mm  woran liegt das ?[dxf](https://we.tl/t-RkeLrxdTtp)  oh man.. sorry. für die ganze fragerei..

Dexus, 2025-03-29:
Bei mir ist es in deepnest, kann es auch nicht nachvollziehen das du andere werte hast

[image]

[image]

[image]

Tigger86, 2025-03-29:
ich hatte den svg Normalizer angeschaltet. jetzt hat es funktioniert.
Danke und ein schönes wochenende

Dexus, 2025-03-29:
Wenn svg preprocessor an ist teste mal mit 96dpi ob das was ändert, bin erst am Montag wieder am Rechner, aber wenn ich das gerade ansehe vom Handy aus, könnte das die Differenz erklären.

claytonchristopher, 2025-03-31:
Thank you for all the hard work to keep this alive. This program is a lifesaver for me.
I do apologize if I am going about this incorrectly, as I have never commented on GitHub.
I too am having problems with import/export scaling. Unfortunately, none of the solutions above work for me. I don't expect a troubleshooting session at all. I just wanted to help by relaying my experience with 1.5.1.
I am using Windows10. Exporting to DXF via Autodesk Inventor 2025. My sample rectangle part is 22.37" x 20.4244". Once imported into Deepnest it measures 0.9" x 0.8". My Deepnest settings are as follows:
Display Units: inches
Space between parts .2
Curve tolerance 0.01
Part rotations: 4
Optimization type: Bounding Box
Rough approximation: false
CPU cores: 8
SVG Normalizer: false
SVG scale: 72
Endpoint tolerance: 0.005
DXF import units: Points
DXF export units: Inches
All other settings are set to default.

I was able to get the import to the correct scale by only changing the following settings, but then export was vastly incorrect.
SVG Normalizer: true
SVG scale: 272
DXF import units: Inches

Should you need aditional information let me know.

Dexus, 2025-03-31:
@claytonchristopher are you able to provide this file, to check what is different.

If you don't like to make it public, you can also send it to stuff (_at_) deepnest.net or direct to me my nickname with same domain.

Regards, Josef

claytonchristopher, 2025-03-31:
[BB14X20 Mold Bottom.txt](https://github.com/user-attachments/files/19541470/BB14X20.Mold.Bottom.txt)

claytonchristopher, 2025-04-01:
First - Not funny.... but kind of funny. I had my wallet out ready to pay.

Second - It works perfectly! Export works as well as long as I remember to scale it to inches in my CAM.
Thank you, Sir, for all your work!

Dexus, 2025-04-01:
First - I am happy about every supporter as a sponsor on GitHub or Patreon.

Second - Pleased to read both 👍🏼

closed 2025-04-02 by Dexus