deepnest-next/deepnest #75: :bug: nest export to dxf not working

trivster, 2025-03-26:
**Describe the bug**

I have changed the file conversion server to "https://converter.deepnest.app/convert" but 'could not contact the file conversion server' error occurs when trying to export a nest as dxf.

**To Reproduce**

Steps to reproduce the behavior:

1. Go to 'Start nest'
2. Click on 'export'
3. Scroll down to 'dxf'
4. See error

**Expected behavior**

A dxf of nested parts should be created

**Desktop**

 - OS: Windows 11 64bit
 - Deepnest.net v1.5.1

Dexus, 2025-03-26:
Same as in your #76 for the debuging Informations.
But thistime, I would ask you for the HAR file with all requests. You can go to the Developer Tools and click to the Network Tab, then in the header is a button to export as HAR.

Please attach it here or upload to wetransfer.com and set a password to it which you can mail to dexus (at) deepnest _DOT(- SPAM prevention)_ net  please use for the email title the `Issue deepnest#75`.

trivster, 2025-03-26:
I will try this later.
In the meantime, I can export as svg and convert using https://cloudconvert.com/svg-to-dxf
The output DXFs are correct scale, but this is obviously not ideal.

trivster, 2025-03-26:
The exporter is now working for me.
I thought the files were empty until I realised that the layer colour is black, but I couldn't see them in the dxf viewer I use because the background is also black!
Thanks for you help

closed 2025-03-26 by Dexus