deepnest-next/deepnest #85: :bug: <title>v1.5.1_nesting doesn't work properly when the number of parts is two or more.

yada886, 2025-04-02:
[image]
[image]
[image]
[image]

**Describe the bug**

 If there is only one part to be placed, the nesting starts without any issues. However, if there are two or more parts to be placed, the screen remains dark and nesting does not start.

1. Go to 'deepnest v1.5.1'
2. make base sheet and import svg file (model)
4. Click on 'start nest'
5. If there is only one part to be placed, the nesting starts without any issues. However, if there are two or more parts to be placed, the screen remains dark and nesting does not start.

**Desktop (please complete the following information):**
 - OS: Windows11 Enterprise
 - deepnest Version: 1.5.1

Dexus, 2025-04-02:
@yada886 please add your `settings.json` located in `%APPDATA%/deepnest`.

closed 2025-04-02 by Dexus

yada886, 2025-04-02:
@Dexus Thank you for your reply.
Please give me more information.
please add your settings.json located in %APPDATA%/deepnest.
→ I cannot find the deepnest folder under %APPDATA%. Where could it be located? Is it acceptable to create a settings.json file within the deepnest folder under %APPDATA%?

Dexus, 2025-04-02:
`C:\Users\<user>\AppData\Roaming\deepnest\settings.json`

yada886, 2025-04-02:
[settings.json](https://github.com/user-attachments/files/19571635/settings.json)
This is my settings.json.

Dexus, 2025-04-02:
@yada886 with v1.5.4 it should be fixed.