deepnest-next/deepnest #89: :bug: <title>

barryfx, 2025-04-03:
**Describe the bug**

DeepNest is not importing the attached SVG file correctly.  It is not showing the outer shape and only imports the holes as separate parts.  This SVG was converted from a DXF file exported from Autodesk Fusion.  Currently nothing happens when attempting to import a DXF.  I saw that the conversion server is down?

**To Reproduce**

Steps to reproduce the behavior:

1. Open DeepNest
2. Import SVG file

**Expected behavior**

One shape should appear in the parts list with holes in it.

**Screenshots**

[image]

**Desktop (please complete the following information):**

 - OS:  Windows 11
 - DeepNest: 1.5.4

**Additional context**

Add any other context about the problem here.

**DEBUG information**

Open DeepNest.  Import the SVG file found below.  (Change from.txt to .svg).  The outer shape does not show up and the inner holes appear as separate parts in the list.  I expected on part to show up.

[image]

[W2.txt](https://github.com/user-attachments/files/19592606/W2.txt)

Dexus, 2025-04-03:
@barryfx thanks for take the time to report the issue.

After a quick check, I see your Path is not closed. Non of them closed. While some get auto closed, the outer shape has a to "big" gap. So once you have closed the path. You should get what you expect.

Regards, Josef

barryfx, 2025-04-03:
That is how it comes out of Fusion.  I'm not sure there is anything I can
do about that.  Thanks for looking into it.

Dexus, 2025-04-03:
Do you export as SVG or DXF? If SVG, you should try DXF or DWG.

Regards, Josef

Dexus, 2025-04-03:
Here your closed SVG: I have also converted it to mm (187,3x18,2mm)

[image]

[Download SVG](https://github.com/user-attachments/assets/14ac38a1-a624-49ff-a9a3-fb7a50076d26)

closed 2025-04-04 by Dexus

Dexus, 2025-04-04:
Yes, that's a bit difficult. I honestly don't know how I can help you with that. You could automate it, but that's not necessarily what you want to do, as it would lead to some completely broken paths.

You could try to customize some of the settings for closing paths, but I'm not sure how much you'd have to go there.

barryfx, 2025-04-04:
I tried importing the fixed svg you sent me and it looks right in the large tab on the right, but the outer path isn't showing up in the list on the left.  The outer path is showing up in red on the right side.  I ran it with debug on and there are no errors in the console.

Dexus, 2025-04-04:
There are some overlapping path, I don't know how you created this in Fusion, but they are intersecting each other, so you dont get the path closed and filled, you can self check it, if you have any vector editor, like inkscape or lightburn or your xTools should also can do it, fill the Path, if it get filled it will work, it you cant fill it with a color, it is a broken/open path.

[image]

[image]