deepnest-next/deepnest #90: Looking for MacOS testers

Dexus, 2025-04-04:
Hi together,

I looking for some MacOS testers, I have build in the last weeks a new Build system, that will build a Mac Apple Store release and also a DMG for outside of the Mac Apple Store, now I need some test candidates, that can test the MAS App via TestFlight and some can test the DMG.

They should remove the problem of the "broken" App, because it was not signed and notarized.

This has now been done, but I don't like to release with the new Build System if we still have problems, so I would like to have your feedback for the new Build System provided app output.

## What needs to be teste?

- The release App via Mac Apple Store - via TestFlight
- The release for non Mac Apple Store releases via DMG file

- I need to know does the App work? Or does it give errors/crashes?
- Is the warning from gatekeeper removed, that the App is untrusted? (DMG/ZIP with deepnest.app)

If you like to be a mac tester, please write a mail to stuff@deepnest.net with the subject "MacOS-Tester" an I will invite you to Apple TestFlight. Also i will mail you the Links to the release files for DMG and ZIP with the App included.

Regards, Josef

RainerPri, 2025-04-14:
Hello Josef i can do some Tests for you.

The actual version did not work on my mac!
If i click on the button's, nothing happens.
See video.

https://github.com/user-attachments/assets/11cee176-8b38-410e-a00b-ab26e823d88f

BR Rainer

jobrissaud, 2025-04-15:
+1 for me as well, running into exactly the same issue than @RainerPri with Deepnest 1.5.4 downloaded from the website, and running on a M1 MBP latest version of MacOS

Dexus, 2025-04-15:
I don't know why, it happends, but this is the problem that a needed lib is missing. 

https://registry.npmjs.org/@deepnest/svg-preprocessor-darwin-arm64/-/svg-preprocessor-darwin-arm64-0.1.10.tgz
https://registry.npmjs.org/@deepnest/svg-preprocessor-darwin-x64/-/svg-preprocessor-darwin-x64-0.1.10.tgz

the *.node file in the tar.gz needs to placed within `deepnest-v1.5.4/resources/app/node_modules/@deepnest/svg-preprocessor` then it should be fixed.

jobrissaud, 2025-04-15:
Odd that it was missing then, I placed the file according to your instructions and it fixed the issue. Thank you!

GarthSnyder, 2025-04-18:
I'd be happy to help with this however I can. However, I'm afraid I've had no luck running the current ARM release (1.5.4) on an M1 Max system. Trying to run it produces the message “deepnest-v1.5.4.app is damaged and can’t be opened. You should move it to the Trash." I tried 1.5.3, but it  behaves the same.

Modern macOS systems notoriously require a whole kabuki ceremony to get unsigned apps running, but this is is something different. I tried to run the executable from a Terminal window, hoping it would at least produce some useful diagnostic messages, but there was no output; the Finder popped up the same "damaged" message.

The ARM version is in fact missing the SVG preprocessor library, but replacing it from the link above doesn't make any difference.

Oddly, the x64 version does run under emulation, although it too is missing the SVG library. As far as I can tell, it's working correctly.

RainerPri, 2025-04-18:
Thank you for that information @Dexus!
This solution fix the problem.

BR Rainer

j0nheck, 2025-04-25:
Happy to test as welll

travellingtechie, 2025-08-30:
Do you still need MacOS testers?

> https://github.com/deepnest-next/deepnest/releases/tag/v1.5.6
> 
> has now the first signed app fpr macos, you can download the ZIP, DMG or PKG. It should now work out off the box, and not cry about "defect" files.

You should update the installation instructions with this information

Dexus, 2025-08-30:
> Do you still need MacOS testers?
>
>
> You should update the installation instructions with this information

Good point I'he make me a note for it