deepnest-next/deepnest #95: Question

stampnlicker, 2025-04-08:
Hi;

First  Dexus thank you for reviving this app, it is great for hobbyists and I am sure you will have commercial fans as well.
I have a question ( I won't say bug, as it is most likely user error.) I have been using 1.5.1 and have it working OK. When 1.5.4 came out I tried to run it with no luck. I installed it and first tried ./deepnest1.5.4 and it returned the error SUID about chrome-sandbox. Th

[image]

[image]

is was expected and I went on to set sudo chown root:root chrome-sandbox and then ran sudo chmod 4755 chrome-sandbox (see screen-grab). I then tried to run it and it loaded but the upper left icon was pink and I could go no further. Have I missed something? 

I am running 
## Software Information:
- **Firmware Version:**                            ADLNV105
- **OS Name:**                                     Ubuntu 24.04.2 LTS
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            X11
- **Kernel Version:**                              Linux 6.11.0-21-generic

stampnlicker, 2025-04-12:
Here is an example Dexus

[image]

stampnlicker, 2025-04-13:
Any chance there is a missing library or dependency, that you could think of?
Thanks

Dexus, 2025-04-14:
You can run `ldd` to check if there are missing libs, but that need to run against each binary within the zip.

stampnlicker, 2025-04-15:
[libvulkan.txt](https://github.com/user-attachments/files/19746041/libvulkan.txt)
[libvk.txt](https://github.com/user-attachments/files/19746044/libvk.txt)
[libGLESv2.txt](https://github.com/user-attachments/files/19746040/libGLESv2.txt)
[libffmpeg.txt](https://github.com/user-attachments/files/19746042/libffmpeg.txt)
[libEGL.txt](https://github.com/user-attachments/files/19746043/libEGL.txt)

Dexus, 2025-04-15:
hmm, don't know why, but i see that the a package is missing: @deepnest/svg-preprocessor-linux-arm64-gnu 

it should be enough to copy the node file to the `deepnest-v1.5.4-linux-x64/resources/app/node_modules/@deepnest/svg-preprocessor` folder

[deepnest-svg-preprocessor-linux-arm64-gnu-0.1.10.tar.gz](https://github.com/user-attachments/files/19752120/deepnest-svg-preprocessor-linux-arm64-gnu-0.1.10.tar.gz)

you can also download it your self if you have node installed via `npm pack @deepnest/svg-preprocessor-linux-arm64-gnu`

Dexus, 2025-04-15:
if you run `deepnest_debug=1 ./deepnest-v*.*.*` you will get a webinspector that shows you what is missing.

Dexus, 2025-04-15:
As I said, it is missing the nodejs module. You can add the download the latest package via npm and copy the *.node file to the directoy.

stampnlicker, 2025-04-15:
Thank you Dexus;
Worked it out. Was a bit late last night and noticed the wrong architecture. Changed to x64 and we have success.

Thank you again for your kind help and patience

closed 2025-04-15 by stampnlicker