deepnest-next/deepnest #96: :bug: export dxf missing unit inforomation

Dexus, 2025-04-08:
**Describe the bug**

The exported DXF missing the unit information and fallback to 0 no units, which also falls back to mm dimension.

tyhwell, 2025-04-22:
export the file into dxf ,the dimension will be Multiplied by 25.4 ,
it is also happen when import dxf ,but not for dwg,----->actual drawing is mm,not inch.
So can we keep the same dimension(size) for it whatever what unit it is

Dexus, 2025-04-29:
> export the file into dxf ,the dimension will be Multiplied by 25.4 , it is also happen when import dxf ,but not for dwg,----->actual drawing is mm,not inch. So can we keep the same dimension(size) for it whatever what unit it is

@tyhwell you use what for settings?

If you use on import (points) and export (inch) your get the right mm value - should also be the same for inch dimensions, so you should not need to change the default settings.

tyhwell, 2025-05-27:
Hello Dexus

Can I propose to add one more setting function as below : (Sorry , I can not find the suggestion block)

Add the dimensions between part to sheet edge, like the dimension A &amp; B
Sometimes we need to change it before the material plan