deepnest-next/deepnest #104: .DXF files when imported do not behave as expected

Enertech-maint, 2025-04-22:
**Describe the bug**

If a .dxf file uses 25.4points to define 1inch there is no setting that will import the .dxf to 1inch.

I use Freecad for drawing. I draw in Imperial units. when FreeCad exports to a .dxf it converts 1inch to 25.4points. (FreeCad is base metric) I also have access to Auto cad lt. Auto Cad Lt draws in points.

</body>

</html>

**To Reproduce**

 I drew a 1X1 square in Autocad, a 1inx1in square in freecad, and a 10mmx10mm square in freecad. Then i imported all 3 file 3 time into deep nest. Each file was imported with the setting "Display Units"  set to "inches" and "DXF import units" set to "Points", then to "inches", then to "mm" and here are the results:

File | Point | inches | mm
-- | -- | -- | --
1inx1in   Autocad | 1.0inX1.0in | 0.0inX0.0in | 0.4inX0.4in
1inx1in   freecad | 25.4inX25.4in | 0.4inX0.4in | 9.0inX9.0in
10mmX10mm   freecad | 10.0inX10.0in | 0.1inX0.1in | 3.5inX3.5in

**Expected behavior**

I had expected when the "DXF Import Units" where set to "mm" the freecad 1inX1in file would have imported 1in X 1in.

**Desktop (please complete the following information):**

 - OS: Windows 10 Pro Version 22H2
 - Browser Chrome
 - Version 135.0.7049.96 (Official Build) (64-bit)
 - Deepnest V1.5.4

Dexus, 2025-04-22:
Each programm is creating a different dxf file, with more or less data. That's the problem. And also the DXF/DWG version are different, so that it is complicated, without "dimensional" definitions to set proper values. But I'm on it and this issue is more or less a duplicate of #96 and other already closed issues.

Dexus, 2025-04-29:
@Enertech-maint can you please share the a file that is 10 inch and 10 mm from both AutoCad and FreeCad?

So I can check the file, and look what is wrong on my side.

Dexus, 2025-04-29:
Thank you, @Enertech-maint it shows exactly what I thought. freecad exports as DXF R12 which has no informations about the dimensions. That means, it is used as "points".

Hope I have already found a way to fix it, will work on it and check within this week.