deepnest-next/deepnest #107: :bug: Could not contact the conversion server

dj1encore, 2025-04-24:
For a completed nest, exporting the nest as a DXF is impossible. An error is received, "Could not contact the conversion server"

sdiverjack, 2025-04-24:
I get this error trying to import a dxf to nest now

Chuzeba, 2025-04-25:
[image]

I'm on this problem too

cletusc, 2025-04-26:
@Dexus Is it possible to have offline DXF conversion? This seems like a major detriment to rely on an external server with no fallback and makes it impossible to use offline.

juanenlozano, 2025-04-28:
I have the same issue too

[image]

CncCumberland, 2025-04-29:
Same issue here, using the version 1.5.4

[image]

Dexus, 2025-04-29:
> Is it possible to have offline DXF conversion? This seems like a major detriment to rely on an external server with no fallback and makes it impossible to use offline.

@cletusc You can code one? I'm a single developer on this project and no full time developer on opensource projects. So If you have the time, contribute in which way you can. 

Currently I can't provicde a DXF2SVG Converter, that allowes me to deliver with deepnest - licenses etc. And toolings which would match the quality and opensource are very rare and/or have a overhead of 5-15GB only for the "converters". (This is right now only windows)

My plan is to support it at somepoint, but unfortunately I can't offer anything off the cuff at the moment.

Dexus, 2025-04-29:
The conversion server is now fully accessible again. Apparently someone has uploaded a ‘buggy’ DXF and tried to exploit the server. 
If there is anything else, just let us know. I have adjusted the monitoring.