deepnest-next/deepnest #119: :bug: The DXF import and export produces wrong size objects.

MetalMusings, 2025-05-07:
The DXF import and export produces wrong size objects.

I have attached a 100x100 rectangle. Exported from Fusion 360 using mm as unit.
Import in deepnest-next.
The size shown in deepnest-next is now 283.4x283.4 mm

[image]
I add a 1000x1000 sheet rectangle, start nesting. export the resulting rectangle.
Import in Fusion 360.
The size of the rectangle is now 7199.111x7199.111 mm

The expected size of the rectangle when imported into deepnest-next was 100x100 mm.
The expected size of the rectangle when imported into Fusion 360 was 100x100 mm

Windows 11 , deepnest 1.5.4

[image]

[image]

Attached the two dxf files.

[100x100-org.dxf.txt](https://github.com/user-attachments/files/20077847/100x100-org.dxf.txt)

[100x100-nested.dxf.txt](https://github.com/user-attachments/files/20077853/100x100-nested.dxf.txt)

Dexus, 2025-05-07:
If you use on import (points) and export (inch) your get the right mm value - should also be the same for inch dimensions, so you should not need to change the default settings.

MetalMusings, 2025-05-07:
Yes that worked. Thanks.

Dexus, 2025-05-07:
#93

closed 2025-05-07 by Dexus