deepnest-next/deepnest #124: :bug: 1.5.5 Failed to Nest - build from source works

1ubuntuuser, 2025-05-13:
**Describe the bug**
The UI works as expected. Pressing the "start nest" button causes the app to start processing but no nests ever show up. 

Building the app from the master git source everything works well.

**Screenshots**

If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**

 - OS: x86_64 GNU/Linux
 - Browser?
 - Version - 1.5.5

**Additional context**

Add any other context about the problem here.

**DEBUG information**

Nothing in console

```
./deepnest-v1.5.5 
{ date: 1970-01-01T00:00:00.000Z, id: '' }
[106720:0513/124121.226294:ERROR:CONSOLE(1)] "Request Autofill.enable failed. {"code":-32601,"message":"'Autofill.enable' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[106720:0513/124121.226358:ERROR:CONSOLE(1)] "Request Autofill.setAddresses failed. {"code":-32601,"message":"'Autofill.setAddresses' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[106720:0513/124121.445431:ERROR:CONSOLE(1)] "Request Autofill.enable failed. {"code":-32601,"message":"'Autofill.enable' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[106720:0513/124121.445618:ERROR:CONSOLE(1)] "Request Autofill.setAddresses failed. {"code":-32601,"message":"'Autofill.setAddresses' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[106766:0513/124123.681303:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
[106766:0513/124123.697577:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
Using local notifications example file
[106720:0513/124124.508784:ERROR:atom_cache.cc(229)] Add _NET_WM_STATE_MODAL to kAtomsToCache
[106720:0513/124124.980905:ERROR:CONSOLE(1)] "Request Autofill.enable failed. {"code":-32601,"message":"'Autofill.enable' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[106720:0513/124124.980960:ERROR:CONSOLE(1)] "Request Autofill.setAddresses failed. {"code":-32601,"message":"'Autofill.setAddresses' wasn't found"}", source: devtools://devtools/bundled/core/protocol_client/protocol_client.js (1)
[106766:0513/124128.027424:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 3 times!
starting background!
```

1ubuntuuser, 2025-05-13:
For context here is the logs  of a working one.
```
{ date: 1970-01-01T00:00:00.000Z, id: '' }
[108892:0513/124652.745937:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
[108848:0513/124655.238008:ERROR:atom_cache.cc(229)] Add _NET_WM_STATE_MODAL to kAtomsToCache
[108892:0513/124655.664380:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
[108892:0513/124658.242903:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 3 times!
starting background!
```

Dexus, 2025-05-13:
What does not Work with which settings? I can run the Linux App on different Distros without a problem.

Here a list for Ubuntu of packages the App need - unfilterd:

```
      "libasound2 | libasound2t64",
      "libatk-bridge2.0-0 | libatk-bridge2.0-0t64",
      "libatk1.0-0 | libatk1.0-0t64",
      "libatspi2.0-0 | libatspi2.0-0t64",
      "libc6",
      "libcairo2",
      "libcups2 | libcups2t64",
      "libdbus-1-3",
      "libexpat1",
      "libgbm1",
      "libgcc-s1",
      "libglib2.0-0 | libglib2.0-0t64",
      "libgtk-3-0 | libgtk-3-0t64",
      "libnspr4",
      "libnss3",
      "libpango-1.0-0",
      "libudev1",
      "libx11-6",
      "libxcb1",
      "libxcomposite1",
      "libxdamage1",
      "libxext6",
      "libxfixes3",
      "libxkbcommon0",
      "libxrandr2"
```