deepnest-next/deepnest #137: Why doesn't a 1000x1000mm box fit into a 1000x1000mm sheet?

semihkaya, 2025-06-19:
Hi,
I'm creating a sheet with dimensions 1000x1000mm and trying to place a box of the exact same size (1000x1000mm) onto it, but it doesn't fit. However, when I slightly reduce the box size to 999.9x999.9mm, it fits without any issue.
Why can't the box fit when both dimensions are exactly 1000x1000mm? What would be the proper approach to handle this situation?

[image]

[image]

[image]

Dexus, 2025-06-19:
I found this bug some days ago, also, I was wondering why 10 100x100 not can storage in 1000x1000 bin.

help would 1000.001x1000.001 to match all into the the bin.

semihkaya, 2025-06-19:
Yeah, I'm running into that exact issue. I can't fit two 500x1000 boxes into a 1000.001x1000.001 space, but I can fit ten 100x1000 boxes. It seems like slightly increasing the "bin" value (the containing area) is the best workaround for these precision issues for now.

radevgit, 2025-11-07:
In my opinion, and I thought a lot about this a while ago:
It is not correct, to expect that ZxZ size part will fit in ZxZ sheet.

From physical point of view, you always have material tolerance both on parts and on sheets.
From numerical point of view (not an ordinary mathematics) the borders have discrete numerical representation.
Having the exact sizes match, means you have overlapping borders, that most collision algorithms as a collision.
That is also not correct.

The usual metal sheet tolerances are 0.01 to even 0.5mm
Not sure, if it make sense to account implicitly for tolerances,
but this is the difference between pure mathematical representation
and real metal works.
[ISO 2768 1 & 2 - ISO General Tolerances Chart (PDF)](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://qualityforum.zeiss.com/migration/images/137_8bb2b6eeb6a9e6554d254c216a890c89.pdf&ved=2ahUKEwi94La1qt-QAxU50AIHHX3FAskQFnoECBAQAQ&usg=AOvVaw2FSJhaZ3bQtlJ4UXhz4Sa8)