deepnest-next/deepnest #149: :bug: Issue with importing dxf files and the imported object scale being incorrect

frr032, 2025-07-24:
I think this is a bug.  If not please suggest a way to resolve my scale issue.

When importing dxf files, the scale is incorrect.  The deepest tool does don't indicate the correct dimensions on the imported list.  After nesting and exporting to a dxf file, the scale is still incorrect.  If i import the same original non-nested dxf files into freecad, the dimensions are correct.

[image]

**To Reproduce**

Import this dfx file.  (change the file ext to dxf)

[025 - Copy.change.ext.to.dxf.txt](https://github.com/user-attachments/files/21399102/025.-.Copy.change.ext.to.dxf.txt)

Nest
Export as dxf
Import the nested dxf into cad
Check the tip to tip dimension of the part.

**Expected behavior**

**Check the tip to tip dimension.  It should be 593.63mm.**

**Screenshots**

CORRECT DIM
[image]

**Desktop (please complete the following information):**

 - OS: win 11
 - Browser edge
 - DN Version 1.5.6

**Additional context**

Dexus, 2025-08-05:
> Hello Dexus, I have just installed the program, and it is working great, but it is converting the dimensions into inches sometimes and vice versa.

Did you change something on the import/export settings? If yes, undo it: inch/points

The problem is that your file, may have no informations of its origin size unit, like inch or mm. If there is no Information about it, it will used - if I'm remember correct - the mm on converting. So your 1200inches get 1200mm or visa versa. 

The main problem is, that what ever I try, i currently don't get a correct conversion. Because the DXF is missing informations to decide the right unit. also the DXF versions making it difficult to have a proper conversation. As not all benefits on the converter are supported. 

But as soon as I have a valide fix for it, it get into a release.

jeranon, 2025-08-05:
> Did you change something on the import/export settings? If yes, undo it: inch/points
>
> The problem is that your file, may have no informations of its origin size unit, like inch or mm. If there is no [...]

This worked for me to get my parts to import correctly. much appreciated!